xodio/xod #1973: Problem with pulse-on-change(byte) and accumulate-string

Cesar-S, 2020-02-19:
### Environment

* Scope: desktop IDE
* Version: 0.33.0

### Steps to reproduce

Try to narrow down your scenario to a minimal working/failing example. That is, if you have a big program causing a problem, start with deleting parts not relevant to the issue and observing the result: is the problem still there? Repeat until you get the most straightforward sequence of steps to reproduce the problem without any noise surrounding it.

### Expected behavior

A patch with changes of byte-type values, which includes, if-else-v, accumulate-string and pulse-on-change (byte).
A flip-flop to verify the pulses.

At the change of value send a pulse and accumulate string

### Actual behavior

At the change of value it sends a pulse and but accumulate-string is blocked.

[image]

nkrkv, 2020-02-19:
From what I see, the behavior is correct: you get a string with the following __six__ characters in it:

- `A` (41h)
- `\x00` (00h)
- `B` (42h)
- `\x00` (00h)
- `C` (43h)
- `\x00` (00h)

The NULL-termination character means “that’s all, guys, thank you” in C++. So if you want to get `ABC` you should filter-out nulls.

closed 2020-02-19 by nkrkv

Cesar-S, 2020-02-19:
Purchased and works correctly, thank .