xodio/xod #1982: Update indicator available in libraries.

Cesar-S, 2020-04-20:
Hi, would an update indicator available in public libraries be possible?

Perhaps an option would be a green icon, as opposed to red when there is an error.

nkrkv, 2020-04-21:
Without a deeper look, I’m not sure the real-time check for new versions is easily possible because it involves network communication at “random” moments. However, at least an explicit action “Check for Library Updates” is possible.

Cesar-S, 2020-04-21:
It should not be automatic, in case you want to use an older version. 

It should not be automatic, in case you want to use an older version.

Thanks.

nkrkv, 2020-04-21:
No-no, I don’t mean the updates should be automatic. It will hurt more than help, for sure. I’m talking about network communication to get metadata and see, are there any updates or not.

Two strategies are possible: implicit (in the background at some time moments) and explicit (on button click or menu hit). The former poses many questions about the “proper” moments. The latter makes things—well—explicit.