xodio/xod #1984: As a novice xoder I want foo so that I qux

Cesar-S, 2020-05-11:
With the DHT11 node and LCD I see that there are still problems.  Although an accelerator node was included, the time is set to 0.2 and the xoders add a clock with 2 or 3 seconds.  Something redundant ...

The time pin should be accessible and be modifiable.

nkrkv, 2020-05-12:
Hello! Are you talking that the boolean `ACT` on `text-lcd-16x2-i2c` is not sufficient and the LCD is missing some updates so that the pulse `UPD` should be brought back?

Cesar-S, 2020-05-12:
Hello, Sorry for not including the links...

Here are two examples, it seems that some have a problem.  

https://forum.xod.io/t/not-reading-temp-from-dht11/3920/3

https://forum.xod.io/t/text-lcd-i2c-16x2-with-dht11-hygrometer-nodes-lcd-err/3721/19

Cesar-S, 2020-05-12:
Now I realize that they are using an old LCD note 🤦‍♂️

nkrkv, 2020-05-12:
Looks like there are plenty of different models of DHT11 in the world. The current version of `dht11-hygrometer` throttles updates with `0.2` seconds intervals which is not sufficient for all models, possibly.

What’s the best option for the `dht11-hygrometer`, what do you think?

- Raise throttling from `0.2` to `1.0`, keep interface the same, hope the things will work for everyone
- Expose the throttling limit as an input pin with `0.2` as default. That is, make the interface more complex
- Expose the throttling limit as an input pin with `1.0` as default.
- Expose the throttling limit as an input pin _and_ replace pulse `UPD` with boolean `ACT`.

Cesar-S, 2020-05-12:
> Expose the throttling limit as an input pin with 0.2 as default. That is, make the interface more complex

In my library I used the throttle node but left the time pin visible.  I think it could be a simple change without affecting the rest.

https://xod.io/libs/cesars/rich-uno-r3/