xodio/xod #2010: "Package Index Broken" error when uploading to Arduino

bxbrenden, 2020-08-04:
### Environment

* Scope: desktop IDE
* Version: 0.34.0
* OS: Debian 10 Buster
* Machine: i5, 16 GB RAM

### Steps to reproduce

0. Build `xod` from source using the [instructions](https://github.com/xodio/xod#building-from-source)
1. Run with `yarn start:electron`
2.  Click `File` -> `New Project`
3.  Double-click the Patch area and use search to add a node of type`xod/common-hardware/led`
4.  `Deploy` -> `Upload to Arduino`
5. There will be a red error popup that reads:
```
Package Index Broken
Error: Unexpected token u in JSON at position 0
Check correctness of the corresponding URL in "/home/<user>/xod/__packages__/extra.txt" and try to update indexes again
```
6. Under the `Board Model` section click `Update` to refresh the list of supported boards
7. The same error mentioned in step 5 will be displayed as a red pop-up.

### Expected behavior

`xod` should refresh the list of supported Arduino devices, parse it, and let the user select one from the drop-down menu.

### Actual behavior

The program cannot parse the JSON returned by the specified URL in `extra.txt`.

The URL in question is:
```
https://storage.googleapis.com/releases.xod.io/packages/esp8266-2.4.3/package_esp8266com_index.json
```

I am able to manually reach that URL, and it returns valid JSON (I validated in python with `json.loads(str)`, and you can do the same with `JSON.parse` in node.js.

[image]

### Console output 

There was one error in the bash terminal, but I do not think it was related.

Here is a screenshot just in case:
[image]

### How to fix 
The URL is valid in the /home/<user>/__packages__/extra.txt` file, and so is the JSON it returns. I also validate the 2 files in the `__packages__` directory, and they can be parsed as JSON. So, I think this may be an issue with reaching out to fetch the resource or the way it is being read.

With a fresh install I was able to fetch and update the list exactly once, but subsequent tries failed. I even deleted the entire directory where `xod` lives and my home directory's `xod` directory and reinstalled. The issue remains.

nkrkv, 2020-08-05:
Hey, thank you for the report. It looks like your version of `arduino-cli` produces output unexpected for XOD (a new version with breaking changes?). Would you share the output of the following two commands?

```bash
arduino-cli version
arduino-cli board listall --format=json
```

brusherru, 2020-08-05:
@bxbrenden try to build it from this branch: https://github.com/xodio/xod/tree/fix-2010-support-arduino-cli-0.11.0 (PR https://github.com/xodio/xod/pull/2012 ).
If the problem persists, please tell us which version of the arduino-cli you are using.

bxbrenden, 2020-08-11:
Thank you both for your quick replies. I will try both of these things tomorrow (building from a different branch and also collecting `arduino-cli` info). Please keep this bug report open in the meantime.

PR #2012 (2020-08-18): Support arduino-cli v0.11.0
It fixes #2010
closed 2020-08-18 by brusherru