xodio/xod #2013: As Enterprise Customer I want to hide some menu items and buttons based on config

brusherru, 2020-08-10:
### Rationale

At the moment we have all menu items and buttons hard-coded in `xod-client`, `xod-client-browser`, and `xod-client-electron`. We want to add the possibility to easily hide some menu items or buttons depending on the config file.

### User story

To hide some menu items, I need to edit `config.json` in the `xod-client` package.

### Acceptance criteria

- [ ] properties in config file effects to menu items visibility
- [ ] config file does not include in the bundle

### How to implement

1. Place a `config.json` in a separate Webpack chunk
2. Import config where it needed (E.G., `xod-client/src/utils/menu.js`)
3. Tweak `items` variable and `submenu`, `onClick` functions from `xod-client/src/utils/menu.js` to avoid rendering hidden menu items.

### How to test

Mock the imported `config.json` with test config that hides some menu items and check does it exists on the page using puppeteer.
To mock the import we probably need to install one new dev dependency, like `testdouble.js`.

For the electron version, probably we need to do something like suggested in this comment: https://github.com/electron-userland/spectron/issues/21#issuecomment-373982282

nkrkv, 2020-08-11:
Please, elaborate on User Story. An example of `config.json` to hide, say,  “Upload to Arduino” and “Update packages” is much desired.

Also, keep Grammarly enabled

nkrkv, 2020-08-12:
Now, the scope of the feature should be defined. It would be quite pointless to hide Edit → Copy, for example. From the input info and some thoughts, the following is the subject to be hidden:

- File → Switch Workspace
- File → Add Library / Publish Library
- Deploy → `<every item>`
- Help → Open Tutorial Project
- Help → Documentation
- Help → Forum

Please, give these action consistent names and add to the issue body.