xodio/xod #2014: As an IDE admin I want to define the list of preloaded libs so that xoders of mine have the same focused environment

nkrkv, 2020-08-11:
For enterprise and educational applications, it is desired to have a list of some libraries preloaded already at the moment when IDE starts. Currently, the list of such libraries is hard-coded and includes only the `xod-*/*` standard libraries. Let’s make it configurable with `config.json` so that an admin could remove the unnecessary libs and add new ones.

### User story

I edit my custom `config.json` and list things to add/remove explicitly:

```json
{
  "libraries": {
    "remove": [
      "xod/waves",
      "xod-dev/*",
      "xod-cloud/*"
    ],
    "add": [
      "xod-dev/servo",
      "mycompany/libfoo",
      "mycompany/libbar@0.2.3"
    ]
  }
}
```

The config above should remove `xod/waves` and any libraries that belong to `xod-cloud` and `xod-dev` except the `xod-dev/servo`. It will also cause the installation of `mycompany/libfoo` at the latest version and `mycompany/libbar` at version `0.2.3`.

The wildcard `*` is only valid in `libraries.remove` and works the same way it does in Bash. A removal entry equal to just `*` is perfectly legal and means “remove everything standard”.

Note, the `remove` key never deletes anything in the workspace. It only instructs XOD IDE which libraries to omit regarding the default list. If a library is already installed in the workspace on the file system it will appear regardless of whether it is listed in `remove` or not.

### Acceptance criteria

- [ ] The `libraries.add` section of `config.json` is taken into account so that new libraries get installed on the launch
- [ ] The new libs are not re-installed if present already
- [ ] The `libraries.remove` section of `config.json` is taken into account so that some default libs (if not present already) are not installed (and thus invisible) on the start
- [ ] The `add` list acts as a mask above `remove` so that if I remove `xod-dev/*` from the default list which includes `xod-dev/servo@0.33.0` and then add `xod-dev/servo` this will leave the requirement for the version `0.33.0` and not the latest version.

brusherru, 2020-08-13:
Should these libraries be saved in a workspace after saving the project?
If so — how they can be managed then, especially for updating stale library versions? 🤔

nkrkv, 2020-08-17:
Clarified that at the end of “User Story”. See “Out of scope” also.