xodio/xod #2016: As enterprise admin I want to tweak an IDE look: change favicon, background image, title rebranding

brusherru, 2020-08-11:
### Rationale

The enterprise admin wants to show their logotype on the background, and show their title of the tab/application.

### Prerequisites

#2013 — introducing a new configuration file

### User story

To brand the IDE admin should update a new section of the `config.json` file:
```
{
  "branding": {
    "title": "XOD Enterpise Edition",
    "noTabs": {
      "backgroundImage": "./my_own_logo.svg",
      "backgroundColor": "#F00"
    }
  }
}
```
Also, the admin should place graphic images in the special directory "branding" in the root of "dist" (or root of the package, which will be copied to the "dist" during the build process). paths to images are specified from this folder, i.e. `./my_own_logo.svg` will be resolved to the `.../xod-client-browser/dist/branding/my_own_logo.svg`

### Out of Scrope

Customizing favicon.ico or application icons, because:
- favicon.ico stored in the service that serves a website, and it much desired to be placed in the root of the public (in case of that search engines will index some images it looks for a default favicon.ico place)
- Admin can change electron application icons only via building a new application bundle

### Acceptance criteria

- [ ] Admin can change the background image and background color of the placeholder when no tabs opened via the config file and placing images to the branding directory
- [ ] Admin can change the title via the config file

### How to implement

1. Replace constant paths or colors in the components with functions which gets values from the config file or fallback to defaults
2. Replace constant title with the function which gets value from the config file or fallback to the default

### How to test

Mock the `config.json` file (like in #2013) and check the title, image paths, background-color via puppeteer.

nkrkv, 2020-08-12:
```
    "noTabs": {
      "backgroundImage": "./my_own_logo.svg",
      "backgroundColor": "#F00"
    }
```

I see no reason to play with the image path. Can we just extract the image out of the `bundle.js` to a `well/known/location/logo.svg` and document the way it can be overwritten?

Also, see no reason to adjust the color. Any color other than current will not place nicely with the surrounding UI. Suggest removing.

nkrkv, 2020-08-12:
> favicon.ico stored in the service that serves a website, and it much desired to be placed in the root of the public (in case of that search engines will index some images it looks for a default favicon.ico place)

It’s our cluster internal detail. How does `yarn start dev:browser` work? It serves static files anyway, so it can serve the favicon. The favicon [is not required](https://www.w3.org/2005/10/howto-favicon) to be on root (although I think we can put in even there). Use `<link>` tag to point to the location.

To put the summary. Customizing favicon at the `xod-client-browser` level is possible in the way similar to `backgroundImage`.

Customizing the icon of desktop IDE is indeed out of scope.