xodio/xod #2019: Add `--split` argument for `xodc transpile`

evgenykochetkov, 2020-08-11:
### Rationale

Currently, generated code is output as a single chunk.
That is undesirable for enterprise users who have custom build systems and some other advanced usage scenarios.

### User story

When called with `--split` argument, for example 

```
xodc transpile foo.xodball main --split -o ./foo/bar
```

`xodс` produces
- a static `runtime.cpp` file
- implementations of used patches, each one in a separate file named like `user__libname__patchname.cpp`
- `main.cpp` — an entry point that ties everything together (contains `runTransaction`, `setup`, `loop`, etc)

and outputs them in a directory specified by the `--output` argument.

knopki, 2020-08-11:
> `main.cpp` — an entry point that ties everything together

Shouldn't the file have a unique name? In the case when several projects are transpiled into one directory with overwriting?

nkrkv, 2020-08-12:
Good starting point but many details are missing or overlooked:

**(1)**

> implementations of used patches, each one in a separate file named like `user__libname__patchname.cpp`

The implementations are potentially C++ template-based, so they can’t be `.cpp`. They have to be `.h`. Anyway, you will refer nodes’ `evaluate` in transactions, so it can’t be just hidden in `.cpp`.

**(2)**

The acceptance criteria are missing. How can we judge the feature is implemented or not? The criteria should include a description of specific steps to successfully consume the artifact.

**(3)**

> xodс produces a static runtime.cpp file

No-no. First, you can’t refer anything hidden in `.cpp`. Second, there are templates that can’t be put in `.cpp`. Third, and most importantly, the current monolith gluing is done because of that single-file requirement; no reason to union the carefully split source files in this mode.

**(4)**

Edge cases not described:

- What if `./foo/bar` does not exist?
- What if `./foo/bar` exists but not empty or not a directory at all?

**(5)**

As @knopki mentioned, how a user should set the main file name?