xodio/xod #2044: Update bug in https://xod.io/libs (C++)

Cesar-S, 2020-09-21:
There is a bug in the C ++ update in web https://xod.io/libs , it still shows a code but the node now includes other nodes, and the main C ++ code is no longer that shown.

example: 
https://xod.io/libs/cesars/rich-uno-r3/rich-touch/

nkrkv, 2020-09-21:
Hello! Thanks for the report. Here’s what I see on the page pointed as the example:

[image]

What’s wrong with it? What C++ code do you expect to see?

Cesar-S, 2020-09-21:
Hi, the node is composed of multiple nodes.

[image]

[image]

Although there is a code, that code is no longer the one shown on the page.

[image]

nkrkv, 2020-09-21:
Ah, I see. So, some obsolete implementation sticks when a node becomes composite.

Marking the issue as a bug.

brusherru, 2020-09-21:
It's a bug on the website, that shows a stale C++ implementation... The code transpiles correctly.
So it's not a core 🤔 
Probably, we have to remove the C++ file at all, but I remember we got some arguments to leave it as is.

closed 2020-09-21 by brusherru