xodio/xod #2075: yarn build error under windows10 and vs2019 community

nklkkrn, 2020-12-13:
Hi guys! On this beautiful night, I was trying to build XOD from the repo. But there was an error that no one expected :)

I would be grateful for any hint how to solve this issue.

So, my environment is Windows 10 x64; Python 2.7.18; VS2019 16.8.3

c:\xod>node -v
v15.4.0

c:\xod>npm -v
7.0.15

c:\xod>yarn
yarn install v1.22.10
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.1.2: The platform "win32" is incompatible with this module.
info "fsevents@1.1.2" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@1.2.8: The platform "win32" is incompatible with this module.
info "fsevents@1.2.8" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
warning " > eslint-import-resolver-webpack@0.6.0" has incorrect peer dependency "webpack@^1.11.0 || ^2.1.0-beta || ^2.1.0".
warning "workspace-aggregator-8289d17b-aba6-4add-8ae6-5d62c3312987 > xod-client-browser > why-did-you-update@0.1.0" has incorrect peer dependency "react@^15.0 || 0.14.x".
warning "workspace-aggregator-8289d17b-aba6-4add-8ae6-5d62c3312987 > xod-client-electron > chai-as-promised@6.0.0" has incorrect peer dependency "chai@>= 2.1.2 < 4".
warning "workspace-aggregator-8289d17b-aba6-4add-8ae6-5d62c3312987 > xod-client > react-codemirror@1.0.0" has incorrect peer dependency "react@>=15.5 <16".
warning "workspace-aggregator-8289d17b-aba6-4add-8ae6-5d62c3312987 > xod-client > react-codemirror@1.0.0" has incorrect peer dependency "react-dom@>=15.5 <16".
warning "workspace-aggregator-8289d17b-aba6-4add-8ae6-5d62c3312987 > belt-holes > @glennsl/bs-jest > jest > jest-cli > @jest/core > jest-resolve-dependencies@24.5.0" has unmet peer dependency "jest-resolve@^24.1.0".
warning "workspace-aggregator-8289d17b-aba6-4add-8ae6-5d62c3312987 > belt-holes > @glennsl/bs-jest > jest > jest-cli > jest-config > jest-resolve@24.5.0" has unmet peer dependency "jest-haste-map@^24.0.0".
[4/4] Building fresh packages...
[6/9] ⠐ @serialport/bindings
[2/9] ⠐ node-sass
[3/9] ⠈ puppeteer
[7/9] ⠈ @serialport/bindings
**error c:\xod\node_modules\node-sass: Command failed.**
Exit code: 1
Command: node scripts/build.js
Arguments:
Directory: c:\xod\node_modules\node-sass
Output:
Building: C:\Program Files\nodejs\node.exe c:\xod\node_modules\node-sass\node_modules\node-gyp\bin\node-gyp.js rebuild --verbose --libsass_ext= --libsass_cflags= --libsass_ldflags= --libsass_library=
gyp info it worked if it ends with ok
gyp verb cli [
gyp verb cli   'C:\\Program Files\\nodejs\\node.exe',
gyp verb cli   'c:\\xod\\node_modules\\node-sass\\node_modules\\node-gyp\\bin\\node-gyp.js',
gyp verb cli   'rebuild',
gyp verb cli   '--verbose',
gyp verb cli   '--libsass_ext=',
gyp verb cli   '--libsass_cflags=',
gyp verb cli   '--libsass_ldflags=',
gyp verb cli   '--libsass_library='
gyp verb cli ]
gyp info using node-gyp@3.6.2
gyp info using node@15.4.0 | win32 | x64
gyp verb command rebuild []
gyp verb command clean []
gyp verb clean removing "build" directory
gyp verb command configure []
gyp verb check python checking for Python executable "python2" in the PATH
gyp verb `which` failed Error: not found: python2
gyp verb `which` failed     at getNotFoundError (c:\xod\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (c:\xod\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (c:\xod\node_modules\which\which.js:80:29)
gyp verb `which` failed     at c:\xod\node_modules\which\which.js:89:16
gyp verb `which` failed     at c:\xod\node_modules\isexe\index.js:42:5
gyp verb `which` failed     at c:\xod\node_modules\isexe\windows.js:36:5
gyp verb `which` failed     at FSReqCallback.oncomplete (node:fs:199:21)
gyp verb `which` failed  python2 Error: not found: python2
gyp verb `which` failed     at getNotFoundError (c:\xod\node_modules\which\which.js:13:12)
gyp verb `which` failed     at F (c:\xod\node_modules\which\which.js:68:19)
gyp verb `which` failed     at E (c:\xod\node_modules\which\which.js:80:29)
[... 77 lines cut ...]
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp verb command build []
gyp verb build type Release
gyp verb architecture x64
gyp verb node dev dir C:\Users\User1\.node-gyp\15.4.0
gyp verb found first Solution file build/binding.sln
gyp verb using MSBuild: C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\15.0\Bin\MSBuild.exe
gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\15.0\Bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64'
gyp info spawn args ]
gyp ERR! UNCAUGHT EXCEPTION
gyp ERR! stack Error: spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\15.0\Bin\MSBuild.exe ENOENT
gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:276:19)
gyp ERR! stack     at onErrorNT (node:internal/child_process:476:16)
gyp ERR! stack     at processTicksAndRejections (node:internal/process/task_queues:80:21)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "c:\\xod\\node_modules\\node-sass\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--verbose" "--libsass_ext=" "--libsass_cflags=" "--libsass_ldflags=" "--libsass_library="
gyp ERR! cwd c:\xod\node_modules\node-sass
gyp ERR! node -v v15.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! This is a bug in `node-gyp`.

nkrkv, 2020-12-14:
Hello! The currently targeted NodeJS is 12, your version is 15. Try to build against NodeJS v12. You can use `nvm` to manage multiple multiple versions of Node side-by-side.

From what I see, you have no Python2 available in your system but it is required to build the `serialport` package from sources. IIRC, in case of NodeJS 12 you are not even required to build `serialport` from sources because it is available as a binary package out of the box. You could also try to stay on NodeJS 15 but install Python 2 and retry but I’m not sure you’ll have no problems at a later step.

nklkkrn, 2020-12-14:
@nkrkv thanks for fast reply!
I`ve tried to do what you say with latest 12 nodeJS (12.20.0), and unfortunately with no success. Of course, I completely removed nodeJS and cleared its caches and other node folders before installation. Then tried installing some other versions of nodeJS 12.*
May be there is a complete list of tools (with specify their versions) for build XOD somewhere in the universe? ;)
PS: When build the sass and serialport nodes bindings in VS2019, there are many erros in nan.*h headers occured.

c:\xod>node -v
v12.20.0

c:\xod>npm -v
6.14.8

c:\xod>python --version
Python 2.7.18

c:\xod>yarn
yarn install v1.22.5
[1/4] Resolving packages...
[2/4] Fetching packages...
info fsevents@1.1.2: The platform "win32" is incompatible with this module.
info "fsevents@1.1.2" is an optional dependency and failed compatibility check. Excluding it from installation.
info fsevents@1.2.8: The platform "win32" is incompatible with this module.
info "fsevents@1.2.8" is an optional dependency and failed compatibility check. Excluding it from installation.
[3/4] Linking dependencies...
warning " > eslint-import-resolver-webpack@0.6.0" has incorrect peer dependency "webpack@^1.11.0 || ^2.1.0-beta || ^2.1.0".
warning "workspace-aggregator-5b725898-c58c-4c76-927c-c0babb7dff61 > xod-client-browser > why-did-you-update@0.1.0" has incorrect peer dependency "react@^15.0 || 0.14.x".
warning "workspace-aggregator-5b725898-c58c-4c76-927c-c0babb7dff61 > xod-client-electron > chai-as-promised@6.0.0" has incorrect peer dependency "chai@>= 2.1.2 < 4".
warning "workspace-aggregator-5b725898-c58c-4c76-927c-c0babb7dff61 > xod-client > react-codemirror@1.0.0" has incorrect peer dependency "react@>=15.5 <16".
warning "workspace-aggregator-5b725898-c58c-4c76-927c-c0babb7dff61 > xod-client > react-codemirror@1.0.0" has incorrect peer dependency "react-dom@>=15.5 <16".
warning "workspace-aggregator-5b725898-c58c-4c76-927c-c0babb7dff61 > belt-holes > @glennsl/bs-jest > jest > jest-cli > @jest/core > jest-resolve-dependencies@24.5.0" has unmet peer dependency "jest-resolve@^24.1.0".
warning "workspace-aggregator-5b725898-c58c-4c76-927c-c0babb7dff61 > belt-holes > @glennsl/bs-jest > jest > jest-cli > jest-config > jest-resolve@24.5.0" has unmet peer dependency "jest-haste-map@^24.0.0".
[4/4] Building fresh packages...
[6/9] ⠐ @serialport/bindings
[2/9] ⠐ node-sass
[3/9] ⠈ puppeteer
[7/9] ⠈ @serialport/bindings
error c:\xod\node_modules\@serialport\bindings: Command failed.
Exit code: 7
Command: prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
Arguments:
Directory: c:\xod\node_modules\@serialport\bindings
Output:
prebuild-install WARN install No prebuilt binaries found (target=12.20.0 runtime=node arch=x64 platform=win32)
gyp info it worked if it ends with ok
gyp info using node-gyp@3.8.0
gyp info using node@12.20.0 | win32 | x64
gyp info spawn C:\Python27\python.EXE
gyp info spawn args [
gyp info spawn args   'c:\\xod\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-G',
gyp info spawn args   'msvs_version=2015',
gyp info spawn args   '-I',
gyp info spawn args   'c:\\xod\\node_modules\\@serialport\\bindings\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'c:\\xod\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\nikol\\.node-gyp\\12.20.0\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\nikol\\.node-gyp\\12.20.0',
gyp info spawn args   '-Dnode_gyp_dir=c:\\xod\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\Users\\nikol\\.node-gyp\\12.20.0\\<(target_arch)\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=c:\\xod\\node_modules\\@serialport\\bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'c:\\xod\\node_modules\\@serialport\\bindings\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\15.0\Bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64'
gyp info spawn args ]
gyp ERR! UNCAUGHT EXCEPTION
gyp ERR! stack Error: spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\15.0\Bin\MSBuild.exe ENOENT
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:268:19)
gyp ERR! stack     at onErrorNT (internal/child_process.js:470:16)
gyp ERR! stack     at processTicksAndRejections (internal/process/task_queues.js:84:21)
gyp ERR! System Windows_NT 10.0.19041
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "c:\\xod\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd c:\xod\node_modules\@serialport\bindings
gyp ERR! node -v v12.20.0
gyp ERR! node-gyp -v v3.8.0

knopki, 2020-12-15:
> gyp ERR! stack Error: spawn C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\15.0\Bin\MSBuild.exe ENOENT

aka "No such file or directory". Looks like problem with `node-gyp` dependencies. According to `node-gyp` [documentation](https://github.com/nodejs/node-gyp#option-1), `windows-build-tools` should be installed and/or `msvs_version` configuration set.

nklkkrn, 2020-12-15:
@knopki thanks! How could I miss it? So many trash messages and some one is important, shit :)
Needed build tools and msvc targeting packs has already installed.
Use `npm config set msbuild_path "C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Current\Bin\MSBuild.exe"` fixed it, but now many new build errors with nan* headers  like these:
  c:\users\User1\.node-gyp\12.20.0\include\node\v8.h(3498): note: see declaration of 'v8::Object::Set' (compiling source file ..\src\serialport.cpp)c:\users\User1\.node-gyp\12.20.0\include\node\v8.h(3498): note: see declaration of 'v8::Object::Set' (compiling source file ..\src\serialport_win.cpp)
and these:
c:\xod\node_modules\@serialport\bindings\node_modules\nan\nan_object_wrap.h(24): error C2039: 'IsNearDeath': is not a member of 'Nan::Persistent<v8::Object,v8::NonCopyablePersistentTraits<T>>' [c:\xod\node_modules\@serialport\bindings\build\bindings.vcxproj]

I'm just amazed this failure. Lazy googling says that need downgrade nodejs to v11 or lower... I haven't had time to test it yet

nklkkrn, 2020-12-15:
With [nodeJS v11.15.0](https://nodejs.org/download/release/v11.15.0/) native modules have been successfully built 👍 
We can conclude that the used native [modules ABI](https://nodejs.org/en/download/releases/) version is 67 :)
Thanks for everyone

closed 2020-12-15 by nkrkv