xodio/xod #2111: Cloud compilation limits exceeded

imjarek, 2021-07-09:
### Environment

* Scope: desktop IDE
* Version: 0.38.0
* OS: Ubuntu 20.04.2 LTS
* Machine: i3, 8 GB RAM

### Steps to reproduce

Because of "Cloud compilation limits exceeded" error for guest user, i attempted to login for a few times, but no avail.
Login button gets inactive after pressing down and nothing happens. Because of this I can give it another try only application restart. I can successfully login to xod.io with these credentials.

### Expected behavior

UI should indicate I'm logged-in or at least show an error (login or password invalid)

### Actual behavior

Nothing happens

[image]

Suanloke, 2021-07-11:
Can't log in.  I think there are some authentication issue at back end.  After logging in, as indicated above, nothing happens.

knopki, 2021-07-11:
Confirmed. Some sort of backend problem.

knopki, 2021-07-11:
Should work now. Feel free to reopen the issue if not.

closed 2021-07-11 by knopki